gopro/gpmf-parser #114: lens parameters of gopro max

CanCanZeng, 2020-10-15:
Hi, I extracted some meta data by using exiftool from _https://exiftool.org/_ 
The command is `exiftool -ee -G3 -v3 -api largefilesupport -m GS010450.LRV > tmp.txt`

I get something like camera intrinsic (lens parameters) like below:

```
GoPro_KLNS = 1081.48076648172 -71.1130226589547 14.9704062180038 -8.043569224[snip]
  | | | | - Tag 'KLNS', type='d' size=40 count=1 (40 bytes, double):
  | | | |   bb2b1a: 40 90 e5 ec 4e 0c 70 10 c0 51 c7 3b c3 63 fa b7 [@...N.p..Q.;.c..]
  | | | |   bb2b2a: 40 2d f0 d9 15 74 52 91 c0 20 16 4e b4 98 f7 6d [@-...tR.. .N...m]
  | | | |   bb2b3a: 3f f1 99 33 82 25 2d 23                         [?..3.%-#]
```
the extracted parameters of front lens are `1081.48076648172 -71.1130226589547 14.9704062180038 -8.043569224 1.0999`

also there are some other parameters like
```
GoPro_CTRX = 4.34998054662002
GoPro_CTRY = -1.8389908280984
GoPro_MFOV = 100
GoPro_SFTR = 1.00002960635123
```

and the transfrom between two lens
```
GoPro_SHFX = 23.1812458157391
GoPro_SHFY = 0.372738888409655
GoPro_SHFZ = 37.0191859120706
GoPro_ANGX = -0.0448620308967071
GoPro_ANGY = 179.946722520132
GoPro_ANGZ = -0.785323833803514
GoPro_CALW = 4056
GoPro_CALH = 3040
```

it seems that GoPro_ANGX GoPro_ANGY GoPro_ANGZ are rotation parameters between two lens, but I cannot guess the meaning of other number

Do you know the meaning of these numbers? I need them to transfrom the dual fisheye video from .lrv file to equirectangular video. Or you can teach me how to realize this in an other way?

paulmelis, 2023-09-02:
Hi @CanCanZeng, did you ever discover the meaning of the Fusion lens and calibration parameters? I'm interested as well, given that Fusion Studio is no longer being developer or supported, and there's no way to produce stitched output on Linux.

CanCanZeng, 2023-09-04:
Not yet.