gopro/gpmf-parser #116: Incorrectly reading sample sizes

lehne, 2020-10-27:
The Apple docs for the format say:

> Sample size
> A 32-bit integer specifying the sample size. If all the samples are the same size, this field contains that size value. If this field is set to 0, then the samples have different sizes, and those sizes are stored in the sample size table.

I have code that writes GPMF, but all the samples are the same size. Thus, that field is set to 0 and the size of the box is only 20 bytes. In the gpmf-parser source, only the second case is handled (differing size samples):

```c
//num is the number of samples, qtsize is the box size. 

	if(num > 0 && qtsize > (num * 4))
	{
	//do something
	} else {
	//close file in error
	}

```

The code only checks that the box size is greater than number of samples times 4 bytes which only happens if samples are differing sizes.

dnewman-gpsw, 2020-10-27:
The MP4 reader has the warning "Way Too Crude MP4|MOV reader", so it does cut corners.  The problem is it is fast and easy to use, we should fix this.  Can you provide a sample file that behaves this way?  And maybe a pull request for the fix?

lehne, 2020-10-27:
So it actually looks like the case is handled but that if statement does not let it run through in that case.
Changing it to:
```c
if(num > 0 && ((qtsize > (num * 4)) || equalsamplesize != 0 && qtsize == 20))
{
```
Seems to work. I can do a pull request.

dnewman-gpsw, 2020-10-27:
I will do it if it is that simple.  All I need is sample media so confirm.

lehne, 2020-10-27:
Let me make a new sample. I had GPS coordinates in my file so I don't want to put that on the internet.

lehne, 2020-10-27:
Here is a sample file.
[3_20201027-144911_0030d4_GoBlue1.zip](https://github.com/gopro/gpmf-parser/files/5447461/3_20201027-144911_0030d4_GoBlue1.zip)

dnewman-gpsw, 2020-11-27:
I believe this is now fixed.  Could you confirm with you own media?

closed 2021-03-01 by dnewman-gpsw