gopro/gpmf-parser #118: Parsing Gyro sensors in timelapse mode

sharon-br, 2020-11-25:
Hey,
I'm trying to parse the gyro sensors from a video I recorded in time-lapse mode in 2 fps.
After extracting all the GYRO data - I see a sample every 0.5 seconds or so - much lower than the actual sensor frequency.
I'm trying to understand what the sample value represents - is it the gyro value at the moment of the sensor capture or is it the sum of all changes over that last 0.5 second?
How should I work with this value?
I'm also having a little difficulty with understanding how to match these samples with the actual offset in the video - the TSMP I see is incremental value from 1 and increments of one on each sample, each sample also contains only 1 value.

dnewman-gpsw, 2020-11-26:
The values are averaged, all values from the previous frame.  They are only slightly useful, not intended for post stabilization or similar.  The down-sampling was needed to reduce the load on the camera, as 200Hz IMU at the longer timelapse intervals would make the metadata track huge.  You should have one sample per video frame.

yakirsudry, 2020-11-27:
Just to make sure I understand. It is just the value of a gyro measurement exactly at the moment a new frame is taken?
Not the properly accumulated rotations throughout the last 0.5 seconds.

So when using timelapse, the IMU measurements are not really usable to stabilize the images? both Accelerometer and Gyro?

Is there any way to get some measurements to stabilize the rotations from frame to frame? because when the camera is carried by a person, his movement might change much faster than 0.5 seconds.

dnewman-gpsw, 2020-11-27:
The ACCL and GYRO values are averaged over the previous 0.5s, not for stabilization. The streams CORI and IORI are designed to be used for stabilization, however there are only valid in TimeWarp, not timelapse.   For some reason the camera can't or doesn't compute the camera orientation in TimeLapse mode.

dnewman-gpsw, 2020-11-28:
VPTS is no longer used, that was on older firmware.  It replacement is the new stream FKSP, used in TimeWarp only, to show the number of frames skipped, allowing you to remap timewarp time to capture real-time.

sharon-br, 2020-11-28:
I'm a bit confused by that since I checked and I'm on the latest firmware of GoPro MAX - 1.50.
I'm also noticing that I do have the CORI stream in timelapse mode - can't it be used?

My plan is to take a video in timelapse mode with a frame every 0.5 seconds - 2 FPS, export it using the GoPro viewer, and extract the frames from it.
Then, for each frame apply the rotation matrix from the quaternions(that I assume follows the regular IMU axis?) that appears in the CORI stream in order to achieve stabilized frames - does that sounds like it should work?

dnewman-gpsw, 2020-11-29:
aha MAX, that should have been in the title.  Okay 1.5 is the latest.  MAX's TimeWarp, is it just Timelapse (with finer steps) if you are using 360 modes.  VPTS pointless, it is just STMP.  Doesn't GoPro Player stabilize your frames, it is using CORI?

sharon-br, 2020-11-29:
True, my apologies for not specifying that.
The GoPro exporter(which I understand is now deprecated) does what I want if I use the world lock option but I want to achieve that myself - does that option uses CORI?

dnewman-gpsw, 2020-11-29:
Yes, CORI is all you needed.  There is one CORI sample per encoded frame.

sharon-br, 2020-11-30:
Ok after some attempts I think I understood what was the issue I had.
First of all - the quaternions from the CORI stream should be initialized in a different order to match the IMU axis - I followed the order in this comment https://github.com/gopro/gpmf-parser/issues/100#issuecomment-656154136

Another issue I had is that I tried to apply the rotation on frames that I extracted from a video that was exported from the GoPro player.
It appears to me that the player applies some kind of effect which messes up frames timing - it tried to smooth up movements which make the CORI values invalid - I couldn't find a way to disable that so I had to extract the frames myself following https://github.com/gopro/gpmf-parser/issues/108 and after applying the CORI values on these frames it seems to work just fine.

Are you aware of this? is there any option to disable the exporter effects? If not I think we can close this issue 😄

dnewman-gpsw, 2020-11-30:
The anti-shake filter is always on. I asked the dev lead if there was a way to turn it off, not currently through the UI.  However an easy(ish) hack using a hex-editor (a lost art), can make the GPMF track invisible to GoPro Player, no stabilization will be applied.  On Windows I use https://mh-nexus.de/en/hxd/  I searched on this string "mhlrmeta" and change it in-place to mhlrmetb"  (it is near the very end of the file.)

sharon-br, 2020-11-30:
That also worked for me, thank you for all your help - I sincerely appreciate all the hard work you are doing here.
I'll close this ticket now - thanks again.

closed 2020-11-30 by sharon-br