gopro/gpmf-parser #127: How to concatenate GoPro MAX .360 video with metadata

secret104278, 2021-01-15:
Hi, I'm trying to concatenate multiple videos of GoPro MAX which are splited due to 8min limits.
I used to achieve this by GoPro player and use World Lock features for single videos. The bad news is that GoPro player will export video with different camera heading for multiple videos, so I can not concatenate the output directly using ffmpeg.
Is there any hint to concentrate multiple .360 directly and form single .360, then use GoPro players to export it and do the world lock?

dnewman-gpsw, 2021-01-15:
You can concatenate the .360 files (with ffmpeg) before conversion.  Yes that works, although I don't remember all the switches required.  GoPro cloud services concatenate all GoPro chaptered media automatically.   Alternatively you can save time and the headache and switch to the new GoPro Labs firmware. See gopro.com/labs  This new firmware can increase the chapter size to 12GB, for longer record times in a single file.

closed 2021-01-15 by dnewman-gpsw

secret104278, 2021-01-18:
@dawilliams-gpsw 
I have tried some ffmpeg commands, however i can't make GoPro Player recognize my video after ffmpeg, and it will only play the first cubemap video track. Do you have any idea how GoPro Player know the video is shoot from GoPro MAX?

dnewman-gpsw, 2021-01-18:
Yes, there is one step.  FFmpeg strips on the 'udta' atom from the MP4 header, you will still need that in the final .360.  I don't know if there is a ffmpeg switch for that. Let me know if you need help on that, as I have a shell tool that copy the udta over somewhere.

secret104278, 2021-01-18:
> Yes, there is one step. FFmpeg strips on the 'udta' atom from the MP4 header, you will still need that in the final .360. I don't know if there is a ffmpeg switch for that. Let me know if you need help on that, as I have a shell tool that copy the udta over somewhere.

@dnewman-gpsw  That will be very help if you can share that shell tool. Thanks befoerehead

I currently use
```
ffmpeg -y -f concat -safe 0 -i ./input.txt \
 -movflags use_metadata_tags \
 -map 0:0 -map 0:1 -map 0:3 -map 0:5 -map 0:6 \
 -map_metadata g \
 -map_metadata 0:s:0 -map_metadata 0:s:1 -map_metadata 0:s:3 -map_metadata 0:s:5 -map_metadata 0:s:6 \
 -c copy -copy_unknown -strict unofficial \
 -tag:2 gpmd \
 -metadata     firmware="H19.03.01.50.00" \
 -metadata:s:  language="eng" \
 -metadata:s:0 handler_name='$(echo -en "\u000bGoPro H.265")' \
 -metadata:s:0 encoder="GoPro H.265 encoder" \
 -metadata:s:1 handler_name='$(echo -en "\u000bGoPro AAC  ")' \
 -metadata:s:2 handler_name='$(echo -en "\u000bGoPro MET  ")' \
 -metadata:s:3 handler_name='$(echo -en "\u000bGoPro H.265")' \
 -metadata:s:3 encoder="GoPro H.265 encoder" \
 -metadata:s:4 handler_name='$(echo -en "\u000bGoPro AMB  ")' \
 -f mov -brand mp41 output.360
```

where `input.txt` is something like
```
file GS038048.360
file GS048048.360
```

* I only map 0, 1, 3, 5, 6 streams because ffmpeg will report error: `non monotonically increasing dts to muxer`.
* I try to use `handler_name='$(echo -en "\u000bGoPro H.265")'` to set `handler_name` according to [this](https://coderunner.io/how-to-compress-gopro-movies-and-keep-metadata/), however I get `handler_name="\f\u000bGoPro H.265"` when ffprobe `output.360`

my ffmpeg version:
```
ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with Apple clang version 11.0.3 (clang-1103.0.32.62)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.3.1-with-options_6 --enable-shared --cc=clang --host-cflags= --host-ldflags= --enable-gpl --enable-libaom --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-libsnappy --enable-libtheora --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-demuxer=dash --disable-libjack --disable-indev=jack --enable-opencl --enable-videotoolbox --disable-htmlpages --enable-libopenh264
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
```

secret104278, 2021-01-18:
@dnewman-gpsw 
I found this will work from [here](https://trac.ffmpeg.org/ticket/6193)

```
mp4extract moov/udta GS038048.360 source-metadata
mp4edit --insert moov:source-metadata output.360 output-withmeta.360
```

At this point GoPro Player can play `output-withmeta.360` as 360 video normally.

secret104278, 2021-01-21:
@dnewman-gpsw  
I found that the Windows GoPro Player can play and export `output-withmeta.360` normally.

However, I found another problem, I merge these 4 videos with `ffmpeg` and `mp4edit`
```
GS010563.360 (A)
GS020563.360 (B)
GS030563.360 (C)
GS040563.360 (D)
```
 to `ABCD.360`

The `World Lock` and `Horizon Level` works, however, when I seek to the range of D, the video is not stabilized. ([video link](https://youtu.be/0-063GPBUfk?t=1440) started from around 24:07 )
If i only merge `BCD` or `CD`, the range D is stabilized correctly.

Furthermore, I've tried GoPro Cloud, it works only when I setup "Auto Upload" from my GoPro MAX camera. The interesting thing is, if we play single chpater like `GS010563.360` in local computer, the video will be stabilized only if we turn on `Horizon Level`, but the completed video generate by GoPro Cloud is always stabilized no matter if we turn on `Horizon Level`. And if we use GoPro Labs, we can record 12G chapter (3x of origin, exactly the number of chapter I can join by ffmpeg), so I'm doubting that 12G/24min is a magic number of GoPro, and is kind of limitation for GoPro to do stabilization.

Do you have any idea about this?

CanCanZeng, 2022-03-23:
Hi, did you solved .360 video concat problem? @secret104278