gopro/gpmf-parser #131: Seeking for clarification on telemetry clock

xshuai1996, 2021-03-30:
Hi,

I managed to parse the meta data stream and found that there were n GPSU segments and n ACCL segments. My questions are:

1. Can I say that 1st ACCL estimation time is the 1st GPSU time? This project: https://goprotelemetryextractor.com/free/# assume it is, but according to the parse result, the 1st GPSU segment was written AFTER ACCL, which intuitively means it recorded the end time of the ACCL block (roughly 200 estimations, i.e. 1 second), instead of start time.
2. Since the total numbers of GPSU and ACCL are the same, the end time or start time of recording telemetry must be missing (otherwise it should be n+1 GPSU). Which is missing, and how can I find it?
3. Is it possible to count the time without the help of GPS (including GPSU)?

The parse log (including every segment) is attached. 

[parse_all.txt](https://github.com/gopro/gpmf-parser/files/6225512/parse_all.txt)

dnewman-gpsw, 2021-03-30:
So you are are not using gpmf-parser code?  It has clock extraction, which uses the mp4 timing along with the timestamps, all of which have been removed from that parse_all.txt example.

xshuai1996, 2021-03-30:
No, I wrote a .py parser since my work required. Could you please give me a general idea about how to extract clock? Do you mean you used Stream #0:2(eng) “GoPro TCD”? I haven't try it because the "GPMF Timing and Clocks" part suggests that different channels have different clock. Also, I think the Java parser I mentioned previously also uses GPSU to count time since the time data is basically same with mine.

xshuai1996, 2021-03-30:
By the way, could you please explain "all of which have been removed from that parse_all.txt example"? Sorry I don't understand. Thank you so much.

dnewman-gpsw, 2021-03-30:
Are looking for time and date or clock extraction for accurate timing for each sample?  If you want time and date, GPSU works if the GPS enabled.  For clock extraction, GPMF uses the timing of the payloads as presented in the mp4. There is some documentation here (see readme) and code to determine the precise timing of the samples within each payload.

xshuai1996, 2021-03-30:
Sorry for confusion. I am new to parsing the video so I'm not familiar with the terminology. Let me explain it again.
-> What I am looking for: I want to figure out the precise time whenever accelerometer record its current state. The real-world time is unnecessary, I simply want the offset (i.e. current - 1st record).
-> My problem about GPSU: ideally I wish the stream to be like this:
    start->...->GPSU->...->ACCL->...->GPSU->...->ACCL->...GPSU->...>end
So that I can calculate the time consumed by each ACCL block using its two neighboring GPSU time. However the actual stream looks like this:
     start->...->ACCL->...->GPSU->...->ACCL->...GPSU->...>end
Thus I don't know how much time the 1st ACCL spents, since there is no GPSU time before it. (I also find some other parsers simply assume the 1st ACCL takes exactly 1 sec, but I doubt the accuracy). That's why I am looking for a "start time" of the stream. 
-> Payload, clock extraction, mp4
I have no idea about what is payload. If clock extraction means "xx year xx month xx day xx:xx:xx", I don't need it. And I don't need the timestamp for each frame. Your documentation indicates the visual data (i.e. video) is in the first track, but I guess it is irrelevant to the sensor.

xshuai1996, 2021-03-30:
An additional time question is, the ffprobe tool shows the duration of my video is 18005.333ms, but the last GPSU time - 1st GPSU time = 16004.999ms. It's reasonable that the first ACCL takes 1000ms (roughly same with others), but there are still approximately 1000ms left, and I have no idea where the GoPro spends on. Do you have any clues?

dnewman-gpsw, 2021-04-03:
Sorry, I don't understand your application.  GPSU is not a precision time, is just a low frequency convenience if GPS UTC time is available, not required for timing. The sample tools here provide accurate timing for each sample in their relation to the video frames within the MP4.

xshuai1996, 2021-04-04:
Cool, I think I understand your idea: it's not a good choice to use GPSU for timing for the accelerometer and gyroscope. Thanks for your patience and help XD

closed 2021-04-04 by xshuai1996