gopro/gpmf-parser #132: Accelerometer orientation

NicolasG3, 2021-04-12:
Hello,
I have extracted GPMF data from several video samples of the project and I always get Z accelerations with a positive sign (like + 9.81) .
I would expect a negative sign (-9.81) since the Z axis points up and the gravity acceleration obviouly points down.  
I used the order and sign defined by ORIN :

- Hero 8 Black ORIN : "zxY"
- Hero 7 Black ORIN: "YxZ"
- Hero 6 Black ORIN: "YxZ"

Am I missing something?

dnewman-gpsw, 2021-04-12:
Positive gravity is correct.  When the camera is in freefall, the accelerometer is zero.  You have to accelerate the camera upward, and positive at 9.81m/s^2 to oppose gravity.  So a stationary camera on a table, on Earth, will have around +9.81m/s^2 on the Z axis.

closed 2021-04-12 by dnewman-gpsw

zhangy76, 2021-07-01:
May I ask why my accelerometer is larger than +9.81m/s^2 at Z axis and not zero at the other two axes?