gopro/gpmf-parser #142: Identify all clips in recording session via gpmf (incl udta gpmf) possible?

blipmusic, 2021-09-15:
Is it possible to somehow automatically identify clips that belong to the same recording session? I'm thinking unique identifiers embedded within the MP4 that link clips. For their legacy camera, Garmin did this by embedding identifiers within the MP4 then logging this identifier in the external FIT-file. I've used that to automatically concatenate clips etc.

If this exists it'd be possible to locate and concatenate both GoPro MP4 and GPMF streams.

Sorry if this is the wrong place to ask, but I can't find a technical forum for the GPMF standard.

I'm trying to write a GPMF parser in Rust from scratch using the documentation in this repo. So far I have successfully used this to automatically extract, concatenate, and parse the mdat DEVC + udta GPMF chunks in the MP4 together with the relative timestamp, derived from the relevant stts, stsz, and stco atoms (the time units in stts seems be plain milliseconds, but perhaps that's just plain luck on my end). A relative timeline is also established for the GPMF stream using these atoms.

AdrianEddy, 2021-09-16:
> I'm trying to write a GPMF parser in Rust from scratch 

I just did exactly that (for mp4 I used `mp4parse-rust`), check out [telemetry_parser](https://github.com/AdrianEddy/telemetry-parser/)

blipmusic, 2021-09-17:
Hey, that's cool! I tried to find something on identifiers that would link clips in your code, but can't see anything?

If it's supposed to be UUID, I haven't found that in any of the MP4-files I have (Hero7 + 9).

Currently I do the parse in two steps, one general, recursive parse, then I try to build filters + processing on top of that. No idea if that's the most effective way. Did it that way for FIT as well.

Unfortunately, I couldn't compile your code on rust 1.55 ([issue](https://github.com/AdrianEddy/telemetry-parser/issues/2#issue-999496550)).

AdrianEddy, 2021-09-17:
I didn't find any indicator of multiple clips in the metadata, there is an extra GPMF box (`extra_gpmf` in `gopro/mod.rs` in my code), it contains some UUIDs but they are the same for different clips from the same camera.

I was just letting you know that Rust GPMF parser exists, but for merging the same clips I guess we have to rely just on filenames. Unless someone from GoPro has some additional information

blipmusic, 2021-09-18:
No worries, I'm over interpreting in hope of finding something. Regardless, I love seeing various solutions for parsing binary data, since I'm quite new at it, so thank you!

Filenames are "ok" , but not ideal. Most people I work with will probably rename and reorganise files before can take a look at them. The intention is to automate where ever possible.

Looking at udta data, "MUID" is logged twice. Once as a "pure" udta field, and again in a udta gpmf stream. The values in the "pure" udta field changes between clips in the same session, but the reinterpreted bytes in a udta gpmf stream (array of UInt32:s) have the same values for clips in the same session. A bit weird, but I'm grasping at straws here and they may be generic identifiers that are actually unrelated to each other. No obvious identifiable sequence at first glance either.

Yes, if a GoPro developer has additional information, I'd be very grateful, even if it's just to confirm that the clips contain no identifier that can be used to determine clip sequence (other than e.g. utc timestamps for the gps).

How does GoPro's own software determine clip sequence and grouping?

dnewman-gpsw, 2021-09-21:
I have looked, and not found a solution beyond the file naming pattern, which is what GoPro software uses. I thought it existed, but it was only used in the early 360 camera products like Omni and Fusion, were each lens was recorded separately (n-chapters, with m-lens, became too much to do without metadata.)

blipmusic, 2021-09-22:
> I have looked, and not found a solution beyond the file naming pattern, which is what GoPro software uses.

Thanks for checking and confirming! I had hopes for the udta data, but no luck. Filename patterns it is.

closed 2021-09-22 by blipmusic