gopro/gpmf-parser #150: Is the imu data smoothed?

CanCanZeng, 2022-04-13:
Hi, I found that the imu has very little noise in still or near-stationary scenes. However, in fast-moving scenarios, the accuracy of imu is not so high. So I guess the imu data has been smoothed, can you confirm that?

dnewman-gpsw, 2022-04-13:
Yes, although I don't know all the details of the filter that is applied. As the camera is performing rolling shutter compensation, where 200Hz Gyro is not sufficient. I believe the original sample rate is around 3200Hz, this data is use in the image stabilization pipeline, and a filtered down-sampled stream is presented to the GPMF track. Only the Fusion camera stored unfiltered IMU data, the large metadata payload slowed some post operations, why we have only 200Hz today.

CanCanZeng, 2022-04-13:
Thanks for your detailed reply, I got it.

closed 2022-04-13 by CanCanZeng