gopro/gpmf-parser #152: the total samples of SHUT data does not equal to the total video frames

CanCanZeng, 2022-04-18:
Hi, I find that for a multi-segment videos data, the total samples of shut data does not equal to the total video frames.

I mean, for a record that longer than 8min, there will be multiple segments. And for the last segment, the total samples of shut data does not match the total video frames. For example, the output of gpmf-parser on one tail-video is :

```
VIDEO FRAMERATE:
  25.000 with 1393 frames
COMPUTED SAMPLERATES:
  ACCL sampling rate = 198.567279Hz (time -0.001778 to 55.797949)",
  GYRO sampling rate = 794.269143Hz (time -0.000087 to 55.799638)",
  MAGN sampling rate = 24.820910Hz (time 0.011295 to 55.811021)",
  SHUT sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  WBAL sampling rate = 8.333333Hz (time -0.040000 to 55.760000)",
  WRGB sampling rate = 8.333333Hz (time -0.040000 to 55.760000)",
  ISOE sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  UNIF sampling rate = 8.333333Hz (time -0.040000 to 55.760000)",
  GPS5 sampling rate = 7.137036Hz (time -0.033645 to 55.731804)",
  CORI sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  IORI sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  DISP sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  GRAV sampling rate = 25.000000Hz (time 0.000000 to 55.800000)",
  WNDM sampling rate = 10.000000Hz (time -0.040000 to 55.760000)",
  MWET sampling rate = 10.000000Hz (time -0.040000 to 55.760000)",
  AALP sampling rate = 10.000000Hz (time -0.040000 to 55.760000)",
```
(1393 video frames)

But when I collect all the shut data, the total samples are 53*26+17=1395 (all but the last payload has 26 samples, and the last one has 17 samples)

My data is recorded by a GoPro Max camera with firmware version : H19.03.02.00.00

Can you confirm if this is a bug for Max firmware v2.0 or on purpose？

dnewman-gpsw, 2022-04-18:
There is no guarantee that the number of video frames, will match the samples for video frame metadata.  It is better that there be more metadata than less.  The video pipeline allows for this, frames that are captured, but may not be encoded, when stop is pressed.  This is considered a minor bug, and it can happen on all our cameras.

closed 2022-04-18 by dnewman-gpsw

CanCanZeng, 2022-04-18:
OK, thanks a lot!

CanCanZeng, 2022-04-19:
Hi @dnewman-gpsw , I also find a .360 data that the two video channel has different frame count, is this also a known bug?
Below is the file info from ffmpeg:
```
{
    "streams": [
        {
            "index": 0,
            "codec_name": "hevc",
            "codec_long_name": "H.265 / HEVC (High Efficiency Video Coding)",
            "profile": "Main",
            "codec_type": "video",
            "codec_time_base": "1/25",
            "codec_tag_string": "hvc1",
            "codec_tag": "0x31637668",
            "width": 4096,
            "height": 1344,
            "coded_width": 4096,
            "coded_height": 1344,
            "closed_captions": 0,
            "has_b_frames": 0,
            "sample_aspect_ratio": "1:1",
            "display_aspect_ratio": "64:21",
            "pix_fmt": "yuvj420p",
            "level": 153,
            "color_range": "pc",
            "color_space": "bt709",
            "color_transfer": "bt709",
            "color_primaries": "bt709",
            "refs": 1,
            "r_frame_rate": "25/1",
            "avg_frame_rate": "25/1",
            "time_base": "1/90000",
            "start_pts": 0,
            "start_time": "0.000000",
            "duration_ts": 9525600,
            "duration": "105.840000",
            "bit_rate": "30019782",
            "nb_frames": "2646",
            "disposition": {
                "default": 1,
                "dub": 0,
                "original": 0,
                "comment": 0,
                "lyrics": 0,
                "karaoke": 0,
                "forced": 0,
                "hearing_impaired": 0,
                "visual_impaired": 0,
                "clean_effects": 0,
                "attached_pic": 0,
                "timed_thumbnails": 0
            },
            "tags": {
                "creation_time": "2017-01-24T15:17:44.000000Z",
                "language": "eng",
                "handler_name": "\u000bGoPro H.265",
                "encoder": "GoPro H.265 encoder",
                "timecode": "15:17:44:05"
            }
        },
        {
            "index": 1,
            "codec_name": "aac",
            "codec_long_name": "AAC (Advanced Audio Coding)",
            "profile": "LC",
            "codec_type": "audio",
            "codec_time_base": "1/48000",
            "codec_tag_string": "mp4a",
            "codec_tag": "0x6134706d",
            "sample_fmt": "fltp",
[... 240 lines cut ...]
                "handler_name": "\u000bGoPro AMB  "
            }
        }
    ],
    "format": {
        "filename": "GS010132.360",
        "nb_streams": 7,
        "nb_programs": 0,
        "format_name": "mov,mp4,m4a,3gp,3g2,mj2",
        "format_long_name": "QuickTime / MOV",
        "start_time": "0.000000",
        "duration": "105.840000",
        "size": "879833825",
        "bit_rate": "66502934",
        "probe_score": 100,
        "tags": {
            "major_brand": "mp41",
            "minor_version": "538120216",
            "compatible_brands": "mp41",
            "creation_time": "2017-01-24T15:17:44.000000Z",
            "firmware": "H19.03.02.00.00"
        }
    }
}
```

dnewman-gpsw, 2022-04-19:
Yes, this is known. Can a frame or two difference.