gopro/gpmf-parser #154: How to handle SHUT

bcardarella, 2022-05-02:
While writing my parser I've come across some FOURCC values that have longer than expected data segments than what their corresponding type seems to want.

In the `hero6.raw` sample file I have the following segment:
[image]

Note `SHUTf` then after the length segment there is a 120 byte value that appears to have repeating 32-bit segments. Are `SHUT`s supposed to be handled in a special way? I see in the README that it relates to shutter speed but I see nothing about it supposed to be a list of shutter speeds

blipmusic, 2022-05-03:
Since there's only one `SHUT` in that sample we should get the same results. I don't see anything odd about it though. The 120 bytes make sense: a base size of 4 bytes, repeated 30 times. Using my own parser I get (ignore NAME and TIME below):
```
SHUT basetype:  102/f, basesize:  4, repeats: 30, pad: 0, NAME: None, TIME: None
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
        [0.008332291]
```

bcardarella, 2022-05-03:
@blipmusic how do you discern that the SHUT is a nesting?

dnewman-gpsw, 2022-05-03:
It is not nesting in GPMF terms, it is simple a stream of potentially changing data.  SHUT 'f' size 4, repeat 30 just means there a 30 floats.  SHUT is exposure time is seconds.  There is typically one entry per frame of video data.   1/0.008332291  means 1/120 shutter speed.

blipmusic, 2022-05-03:
`SHUT` is not a nested stream, due to the type not being 0. I.e. it does not contain more streams, but data values. All nested streams (`DEVC`, `STRM`) have type 0. Basically my strategy is to check which type any stream is before attempting further parsing. If type 0, I call my parsing algorithm recursively, until the initial type check find a non-zero type value.

bcardarella, 2022-05-04:
So if the `repeat` value is greater than `1` is it always assumed to be an array

bcardarella, 2022-05-04:
@blipmusic I think that's a good approach, in Elixir I can just generate charlists which should be very performant. Thanks!

closed 2022-05-06 by bcardarella