gopro/gpmf-parser #155: Metadata sync (again)

time4tea, 2022-05-03:
Hi - I am trying to sync metadata with Video frames, and struggling a bit. I can see that in https://github.com/gopro/gpmf-parser/blob/151bb352ab3d1af8feb31e0cf8277ff86c70095d/GPMF_utils.c#L31 there is a function to estimate the video/metadata sync/offset.

For e.g. HERO9, with a timestamp field, this seems to work ok. However in older cameras the STMP isn't available. I was wondering how accurately the two sync estimation functions worked.

For example, for a given HERO9 track, the sample rate for GPS5 is:
```
COMPUTED SAMPLERATES:
GPS5 first sample at time -7.017ms, last at 707678.489ms
  GPS5 sampling rate = 18.112565Hz (time -0.007017 to 707.678489)",
```

However, by adding a `rate = 0.0; usedTimeStamps = 0;` at line 189, just before: https://github.com/gopro/gpmf-parser/blob/151bb352ab3d1af8feb31e0cf8277ff86c70095d/GPMF_utils.c#L189 so to force the estimation function used in older cameras, the exact same track reports:

```
COMPUTED SAMPLERATES:
GPS5 first sample at time 1939.776ms, last at 708028.260ms
  GPS5 sampling rate = 18.153532Hz (time 1.939776 to 708.028260)",
```

This gives approximately a two second difference in the start time of the metadata track. This is quite a difference. 

I'd appreciate any guidance in syncing the metadata and video for older cameras. 

Thanks!

dnewman-gpsw, 2022-05-03:
It is more accurate on older cameras, and not needed on the newer cameras.  The increased difference is why timestamps were added on newer cameras as internally the video image data is delayed compared with the metadata -- this relates to the Hypersmooth image processing.  In the pre-HS cameras, data arrive proximately together, so the computed sync make sense.

closed 2022-05-03 by time4tea

tve, 2022-05-04:
> this relates to the Hypersmooth image processing

Can you clarify a little the models? One of the tracks we're looking at comes from a Hero 7 black with hypersmooth. The recording was started from cold, i.e., the camera didn't have a GPS lock. The video track starts immediately, the metadata track starts several seconds later, and the GPS lock (and thus GPS timestamps) comes yet later in the metadata track. The Hero 7 does not produce STMP.

dnewman-gpsw, 2022-05-04:
It was with HERO7 that the issue was first observed and fixed with HERO8 by adding timestamps so that metadata is presented with the correct video frames.   On HERO7 the data in GPS is about 1 second misaligned.  It is consistant, so it can be compensated for.