gopro/gpmf-parser #156: 32-bit float data from hero6.raw

bcardarella, 2022-05-06:
Towards the end of the hero6.raw sample file there are a few chunks of data that identify of type `'f'` for a 32-bit float but the size is 12 with a repeat of 30. For a 32-bit float the size should be 4. Or is there something I'm missing here?

bcardarella, 2022-05-06:
I'm actually, seeing this for other types as well. The `s` type which should be a 16-bit signed integer, so 2 bytes has a size of 6 instead with a repeat of 204.

dnewman-gpsw, 2022-05-06:
This is correct, as it describes the array size.  So float [][3] would be type 'f' with a size 12.  short [][3] is type 's' with size 6.

bcardarella, 2022-05-06:
@dnewman-gpsw ah, so an array of arrays?

dnewman-gpsw, 2022-05-06:
Yes.  The repeat is the number on time entries.   So GPMF is always in form   type[repeat][typesize/sizeof(type)]   e.g. float[repeat][typesize/sizeof(float)]

bcardarella, 2022-05-06:
What about char values? Do those remain as two-dimensional array?

dnewman-gpsw, 2022-05-06:
No 1D, generally you will typesize = strlen("the string"), and a repeat of 1.  On the earliest cameras this was reverse, but still 1D.  The only common acception, but still following or the rules.  SUIN/UNIT fields can have a 2D array of chars, which is 1D arrary of fixed length strings.

bcardarella, 2022-05-06:
@dnewman-gpsw I'm seeing in the `hero6.raw` file that I get `['c', 1, 0, 50]` for the length segment. In this case I was reading this as a <type><size><repeat> but your comment is saying the order should be reversed? Or is the hero6 of that older format?

blipmusic, 2022-05-09:
@bcardarella See this thread for ideas:
https://github.com/gopro/gpmf-parser/issues/143

I "convert" all 'c' streams as if it had repeats 1 before parsing as a string (switching place of size and repeat if necessary).