gopro/gpmf-parser #157: DEVCs

bcardarella, 2022-05-08:
Parsing metadata taken directly off my Hero Black 10 and I have 710 DEVCs all from the same device. Are these supposed to be composed into a single key? Or do they serve another purpose all being separate?

dnewman-gpsw, 2022-05-08:
These are just timed payloads, they technically could be all clumped into one, but that is not how video capture works.  Video is not one payload, audio is not one payload.  DEVC is just the first tag in a GPMF metadata payload.

closed 2022-05-08 by dnewman-gpsw

bcardarella, 2022-05-08:
@dnewman-gpsw is each instance of the DEVC a certain timed offset from the video start? I'm missing how I can determine which datapoint in the metadata payload is aligned for a particular timestamp of the video.

dnewman-gpsw, 2022-05-08:
That is in MP4 index, each payload has a time, this is how video and audio sync is restored, same for metadata.  So every DEVC payload has a timestamp in the MP4 index.  For NTSC video captures, most GPMF payloads will have the timing 0, 1.001, 2.002 etc, to match the videos 29.97/59.94 timing, for each video keyframe there is a matching DEVC (GPMF payload).  For 30p video the metadata for the first 30 frames, are stored in the first GPMF payload (timing 0 to 1.001s.)  This is a simplified explanation of MP4 / MOV timing. Precision timing of each individual stream in GPMF is more complex. The time-stamps (STMP) for metadata describing each video frame, data like SHUT, EISO and CORI, are a way to bridge between the MP4/MOV domain and that of each GPMF stream. As the github readme explains (section GPMF Timing and Clocks), timing is hard with clocks that are independent, STMP is a the camera system clock, independent to video clock, indepentent to IMU clock, etc., but you can use the STMPs to content on all the timing to together for sub-frame, high precision. The only reason tools like ReelSteady can work, is precision timing can be extracted after capture.