gopro/gpmf-parser #159: gpmf_demo.c with time-lapse SEGV / program incompatible?

time4tea, 2022-06-06:
Hi,

I'm trying to show the timing and packet information from a time-lapse recording. The recording does appear to have a data stream.

However, the short recordings I have cause a SEGV when using the gpmf_demo.c

The SEGV is caused by a problem in cleanup, with the following sequence of events:

https://github.com/gopro/gpmf-parser/blob/master/demo/GPMF_mp4reader.c#L895

for the time lapse videos, sample count is always 1, while `num` seems to be the number of frames or something, so always greater than 1.

This causes the loop of `entries` to exit, without setting `mp4->basemetadataduration`

The program continues to:

https://github.com/gopro/gpmf-parser/blob/master/demo/GPMF_demo.c#L144

calling: 

https://github.com/gopro/gpmf-parser/blob/master/demo/GPMF_mp4reader.c#L1007

which exits with an error condition, jumping to `cleanup`, which SEGV, in GPMF_FreeCodebook due to ms->cbhandle being improperly initialized in this code path (although this is hard to see because metadata_stream is used uninitialised.

https://github.com/gopro/gpmf-parser/blob/master/demo/GPMF_demo.c#L63

maybe consider something like

`GPMF_stream metadata_stream = {0}, ...`

I'm trying to do this to show timing information for meta data streams in the time-lapse footage.

Thanks!

dnewman-gpsw, 2022-06-07:
That was bug fixed in the develop branch, I just moved that fix over.

closed 2022-06-07 by dnewman-gpsw

time4tea, 2022-06-11:
Awesome. Thanks for merging the fix.