gopro/gpmf-parser #163: Hero 8 Black FACE definition

stevenh, 2022-07-05:
The README specifies the definition of `FACE` Hero 6 and Hero 7 but Hero 8 seems to be a third variant with the complex type definition of `Lffffff`.

Could you document what this type is defined as, specifically the additional 2 fields?

stevenh, 2022-07-07:
Thanks @dnewman-gpsw could you confirm the definition for Hero9, I'm guessing `BLffffff` ?

Also do you have a sample file for hero9 as that would be great for testing against.

dnewman-gpsw, 2022-07-08:
Sorry, I made an error on the readme. That is for HERO10, not HERO9.  HERO9 and HERO8 have the same FACE metadata.  HERO10 uses the more compact "BBSSSSSBB", switching from floats to unsigned shorts or bytes (28 bytes down to 14 bytes per FACE).  It also adds a blink detector.

struct face_data {
  byte version; //4
  byte confidence; //0-100
  uint16_t face_id;
  uint16_t face_x; //0-65535
  uint16_t face_y; //0-65535
  uint16_t face_w; //0-65535
  uint16_t face_h; //0-65535
  byte smile_percent; //0-100
  byte blink_percent; //0-100
};

Example:
    "STRM": {
      "STMP": 2039432,
      "TSMP": 45,
      "STNM": "Face Coordinates and details",
      "TYPE": "BBSSSSSBB",
      "SCAL": [1, 1, 1, 65535, 65535, 65535, 65535, 1, 1],
      "FACE": [4,52,0,10376,17991,19011,39563,0,98],
      "FACE": [4,99,1,17322,11438,16759,36135,0,99],
      "FACE": [4,87,1,18841,11954,14813,36529,0,94],
      "FACE": [4,97,1,19199,12591,14762,35953,0,92],
      "FACE": [4,99,1,21469,12924,12748,35953,0,90],
      "FACE": [4,99,1,21503,13076,15854,36377,0,92],
      "FACE": [4,99,1,23807,12864,15786,36590,0,95],
   }

stevenh, 2022-07-08:
Awesome thanks @dnewman-gpsw!