gopro/gpmf-parser #165: Understanding ORIN/ORIO/MTRX for parsing ACCL/GYRO

time4tea, 2022-08-05:
Hiya - I'm just trying to understand parsing ACCL and GYRO in a way that is compatible across device versions.

I guess I have two questions:

1. I don't really understand the need for MTRX - if i have a struct of x,y and z - why do I need more than ORIN in order to parse it from the ACCL - this tells me the order and sign of the three components. Is that right? I feel I must be missing something fundamental.
2. In hero5 and 6 and Max (maybe others), there is no ORIN information. Is the "default" ORIN therefore "ZXY" - and will any other devices have an ORIN in the ACCL/GYRO, or is some other a-priori knowledge needed to imply some ORIN (e.g use the DVNM to load a known ORIN value for that device model - although many use just "Camera")

Thank you!

dnewman-gpsw, 2022-08-06:
The original GPMF camera, HERO5, used ZXY without reporting any metadata on the orientation. We have since attempted to preserve this.  However, hardware engineering kept moving the IMU, so the actual orientation changes between camera models. So a camera with ORIN and ORIO has recorded the stored orientation, but also noting the preferred output orientation (to match HERO5.)  If you are building your own parser, use ORIN to know the way the raw data is stored.  If you are using gpmf-parser to extract developed output, ORIO is the way it is presented. So if you use gpmf-parser as is, all GoPro cameras will output ZXY, no matter the orientation of the IMU in the camera.  MTRX can be ignored as redundant, it reflects the same orientation change as ORIN to ORIO.  It was never used as intended to store orientation calibration.  In later cameras, as the CPU performance improved, all the orientation corrections and calibrations are applied to the ACCL and GRYO data, so these cameras only have a ORIN for completeness.

time4tea, 2022-08-06:
Thank you. That is really useful! Just to check one last detail - you say later cameras will correct ACCL orientation... but they still have an ORIN? So does this mean that the ORIN needs to be ignored, as the translation has already been applied? Not quite sure I completely.follow that last bit. Thank you for the clarifications!

dnewman-gpsw, 2022-08-08:
They will have ORIN for completeness, so the orientation is documented.  It doesn't need to be ignored, but it is passive.  Only when you have a different ORIO does the orientation get processed.