gopro/gpmf-parser #166: Lower GPS frequency on HERO10 01.46

JuanIrache, 2022-08-30:
Hi.

I'm seeing a lower frequency for GPS data in the latest HERO10 firmware. About 10Hz.

Is that intentional? A bug?

Thank you.

dnewman-gpsw, 2022-08-30:
It is intentional. I believe it is to reduce power consumption.

JuanIrache, 2022-08-31:
Thanks for your reply. I'm not sure this is the right place for feedback, but in case it's useful: I think users that care about GPS data are generally happy to save power in other ways: disable other features, carry extra batteries, plug a power bank... Users that don't use the GPS can disable it completely, and the recently added "Video Performance modes", include a Battery Saving mode where disabling the GPS and other non-essential features makes sense.
If this decision is final, and especially if it affects upcoming camera models, I hope at least the Labs firmware gets a toggle for full GPS data.

dnewman-gpsw, 2022-08-31:
I agree, and hope to add this control for Labs.

JuanIrache, 2022-09-18:
Hi again. My tests comparing the old and new firmwares (18Hz vs 10Hz GPS) seem to indicate a (VERY SLIGHT) improvement in GPS stability (finding a valid signal sooner, keeping a usable signal in difficult circumstances...) with the new firmware. I haven't done enough tests to guarantee the improvement is there, and they are by no means scientific.

I have alseo been reading the data sheet of the GPS module, and it turns out it can only do 18Hz when connected to just one GNSS constellation (say GPS only). If using multiple constellations (say GPS+GLONASS+Galileo), which would help a lot with signal stability by increasing the number of available satellites, it does 10Hz. I see the GPS chip was changed at some point during the HERO10 production, but both models share this feature: 10Hz for multiple constellations, 18Hz for one.
Ublox UBX-M8030 https://content.u-blox.com/sites/default/files/products/documents/UBX-M8030_ProductSummary_%28UBX-15029937%29.pdf
Ublox MAX-M10S https://content.u-blox.com/sites/default/files/MAX-M10S_DataSheet_UBX-20035208.pdf

Could that be a nice side effect (or the core reason) why the GPS frequency was lowered?

dnewman-gpsw, 2022-09-18:
I think you are correct.  All the notes I can find internally the 10Hz switch enables the concurrent GNSS mode (multi system).  I don't know for sure, seems likely.  Time to lock, is something we are always working to improve.

realchrisolin, 2022-09-20:
given the topic at hand, is there anyway for consumers running the labs firmware to log NMEA/UBX debug messages to a text file? Has GoPro done any tests on what causes TTFF delays? Is the receiver being warm or cold started during power on/any relevant details worth knowing? Also, the M8030-CT supports 3 concurrent constellations but appears neither Galileo or Beidou are used to get a fix. Is that intentional?

dnewman-gpsw, 2022-09-20:
Now we are way beyond the scope of this projects.  Sounds like a feature required for GoPro Labs.

realchrisolin, 2022-09-20:
agreed, that's why I asked if it was available for consumers running the labs firmware. It's been over a year since the H10 was launched and the GNSS logging rate change was just recently made. GNSS reception has always been unreliable in my experience with my GoPro's (I'm not alone either). If your engineering team is so busy that it took ~11 months to find something that was spelled out in the receiver docs, you have capable customers that are willing to help find what additional improvements can be made.

Since the receiver appears to support three concurrent constellations, I'm a mildly frustrated as a paying customer/power user that Galileo isn't enabled after this snafu came to light. Hopefully you can get this relayed to the appropriate people.

drcrallen, 2022-09-29:
I've had a few videos half-ruined because I didn't wait for the GPS to stabilize before recording. For me a faster power-on-to-GPS time will save more videos than going back to 18Hz. Also loosing video due to Hero10 overheating has been an issue when in a wind-protection screen is on the camera. So I am interested in the reduced power consumption to keep it from overheating.

As a counter-point to what is mentioned elsewhere in this thread, I am excited to try the 10Hz data in the systems I am using (RaceRender3) because the benefit can easily outweigh the impact of reduced frequency.