gopro/gpmf-parser #171: FACE id changes constantly in version 4

JuanIrache, 2022-11-25:
Hi. I have recently been experimenting with the FACE stream on HERO10 cameras. With previous versions, the recorded ID seems to be consistent with the number of people in the frame, but with the HERO10 version (ver,confidence %,ID,x,y,w,h,smile %, blink %) the ID number seems to be increasing every few samples. Could I be interpreting something wrong?

Here's a snippet of what I see. The ID should be the third value in the array. I can provide the original file privately.
```
[
{
"STMP": 9443744,
"TSMP": 130,
"STNM": "Face Coordinates and details",
"TYPE": "BBSSSSSBB",
"SCAL": [1,1,1,65535,65535,65535,65535,1,1],
"FACE": [
[[4,100,3422,24114,27596,52259,19516,60,0]],
[[4,100,3707,31602,26333,56610,19513,57,0]],
[[4,99,3711,32754,26214,26149,19514,58,0]],
[[4,99,3712,32793,25975,30502,19515,59,0]],
[[4,100,3710,32434,25940,21542,29499,59,0]],
[[4,100,3710,32293,26060,52261,26170,58,0]],
[[4,100,3709,32165,26214,26148,58938,58,0]],
[[4,100,3713,33087,26418,12831,62265,57,0]],
[[4,100,3713,33074,26589,56607,35896,56,0]],
[[4,100,3713,33189,26777,39198,65335,55,0]],
[[4,100,3714,33407,26930,12830,26167,55,0]],
[[4,100,3715,33714,27084,52253,62263,55,0]],
[[4,100,3716,33983,27203,17181,55607,55,0]]]
},
{
"STMP": 10483744,
"TSMP": 143,
"STNM": "Face Coordinates and details",
"TYPE": "BBSSSSSBB",
"SCAL": [1,1,1,65535,65535,65535,65535,1,1],
"FACE": [
[[4,100,3717,34213,27340,52253,55606,54,0]],
[[4,100,3713,33266,25428,21538,9790,62,0]],
[[4,100,3714,33369,25514,43554,3135,63,0]],
[[4,100,3714,33458,25616,4129,65336,56,0]],
[[4,100,3714,33445,25753,39201,65336,56,0]],
[[4,100,3714,33394,25684,21538,16184,56,0]],
[[4,100,3714,33292,25736,34847,9784,56,0]],
[[4,100,3710,32319,25736,34849,55608,56,0]],
[[4,100,3709,32217,25650,12833,62264,56,0]],
[[4,100,3709,32153,25548,52257,65336,56,0]],
[[4,100,3709,32127,25514,43553,58936,56,0]],
[[4,100,3709,32140,25582,60961,52280,56,0]],
[[4,100,3709,32140,25684,21537,48951,55,0]]]
},
{
"STMP": 11523744,
"TSMP": 156,
"STNM": "Face Coordinates and details",
"TYPE": "BBSSSSSBB",
"SCAL": [1,1,1,65535,65535,65535,65535,1,1],
"FACE": [
[[4,100,3709,32127,25821,56609,48952,56,0]],
[[4,100,3709,32178,25906,12833,42552,56,0]],
[[4,100,3709,32153,25958,26145,39223,55,0]],
[[4,100,3709,32178,26179,17185,35895,55,0]],
[[4,100,3709,32140,26231,30497,32567,55,0]],
[[4,100,3709,32089,26282,43553,39223,55,0]],
[[4,100,3709,32025,26333,56609,32567,55,0]],
[[4,100,3708,31973,24575,65313,29502,62,0]],
[[4,100,3708,31884,24575,65313,39231,63,0]],
[[4,100,3708,31845,24609,8481,42559,63,0]],
[[4,100,3708,31820,24643,17181,35902,62,0]],
[[4,100,3708,31820,24609,8477,22846,62,0]],
[[4,100,3708,31820,24575,65309,32574,62,0]]]
}
]
```

Thank you

dnewman-gpsw, 2022-11-25:
That is clearly not working correctly.  However it did not happen in the quick test I did here.  So factory reset the camera, and try again.  If repeats, maybe it is an issue with a particular video mode.

JuanIrache, 2022-11-25:
I switched to 1080P 25fps, disabled all the fancy stuff (hypersmooth, high bitrate) and got the same result, both on Labs firmware and fw versions 1.42 and the latest 1.46. The issue seems to be there. Could I be reading the data wrong? I remember the FACE data in the HERO7 was a bit tricky, as th array had a bunch of empty values "unuse[17]" before the smile.

Here's one of my test files (just the raw gpmf data, the video is a single face): https://www.dropbox.com/s/6jfrer4wniqk33o/GL014218.bin?dl=0

dnewman-gpsw, 2022-11-26:
HERO7 Face data was a screw-up :)

Seems you are have parsing issues.  The ID will always start at zero, and only increase with each new face detected.  This is the output from one of my tools (first payload)

    "STRM": {
      "STMP": 43443,
      "TSMP": 13,
      "STNM": "Face Coordinates and details",
      "TYPE": "BBSSSSSBB",
      "structure (SCAL)": "1-axis 9 samples of type 'S' (UNSIGNED_SHORT)",
      "SCAL": [1, 1, 1, 65535, 65535, 65535, 65535, 1, 1],
      "complex structure (FACE)": "14-byte samples",
      "FACE": [4,99,0,29559,18507,8191,18689,0,0],
      "FACE": [4,100,0,29559,18507,8191,18689,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,18628,0,0],
      "FACE": [4,100,0,29559,18568,8191,20813,0,0],
      "FACE": [4,100,0,29559,20752,8191,18689,0,0],
      "FACE": [4,100,0,29559,20752,8191,18689,0,0],
      "FACE": [4,100,0,29559,20752,8191,18689,0,0],
      "grouped data (FACE)": "13 samples"      
    },

ID is 0 for most of the samples, the highest it got to was 8.

"TYPE": "BBSSSSSBB" which is a structure like this:

struct {
  uint8_t  version;
  uint8_t  confidence;
  uint16_t  ID;
  uint16  facebox[4];
  uint8_t  smile;
  uint8_t  blink;
}

JuanIrache, 2022-11-26:
Thank you. I will review my process

closed 2022-11-26 by JuanIrache

JuanIrache, 2022-11-26:
It was indeed a bug in my parsing process. All good now. Thanks.

Blinks seem to half-fire constantly in some shots. Smile and everything else works great! https://youtu.be/MeMIK2KVz3A