gopro/gpmf-parser #172: the cliped .360 file has different stream order, how to reorder them

CanCanZeng, 2022-12-08:
Hi, I don't know if it is suitable to ask question here, but any suggestion is appreciated.
I find that the cliped .360 file from GoPro Player has different stream order, how can I reorder it as normal .360 file?

the output of `ffmpeg -i a.360` of normal .360 file  is
```
Stream #0:0(eng): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt709), 4096x1344 [SAR 1:1 DAR 64:21], 30036 kb/s, 25 fps, 25 tbr, 90k tbn, 25 tbc (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro H.265
      encoder         : GoPro H.265 encoder
      timecode        : 07:07:28:03
    Stream #0:1(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 189 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro AAC  
      timecode        : 07:07:28:03
    Stream #0:2(eng): Data: none (tmcd / 0x64636D74) (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro TCD  
      timecode        : 07:07:28:03
    Stream #0:3(eng): Data: bin_data (gpmd / 0x646D7067), 88 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro MET  
    Stream #0:4(eng): Data: none (fdsc / 0x63736466), 18 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro SOS  
    Stream #0:5(eng): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt709), 4096x1344 [SAR 1:1 DAR 64:21], 30020 kb/s, 25 fps, 25 tbr, 90k tbn, 25 tbc (default)
    Metadata:
      creation_time   : 2022-05-25T07:07:28.000000Z
      handler_name    : GoPro H.265
      encoder         : GoPro H.265 encoder
      timecode        : 07:07:28:03
    Side data:
      displaymatrix: rotation of nan degrees
    Stream #0:6(eng): Audio: pcm_s32le (in32 / 0x32336E69), 48000 Hz, 4.0, s32, 6144 kb/s (default)
```

and the output of cliped file is
```
Stream #0:0(eng): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt709), 4096x1344 [SAR 1:1 DAR 64:21], 30083 kb/s, 25 fps, 25 tbr, 90k tbn, 25 tbc (default)
    Metadata:
      creation_time   : 2022-05-25T07:08:15.000000Z
      handler_name    : GoPro H.265
      encoder         : GoPro H.265 encoder
      timecode        : 07:10:31:13
    Stream #0:1(eng): Video: hevc (Main) (hvc1 / 0x31637668), yuvj420p(pc, bt709), 4096x1344 [SAR 1:1 DAR 64:21], 30008 kb/s, 25 fps, 25 tbr, 90k tbn, 25 tbc (default)
    Metadata:
      creation_time   : 2022-05-25T07:08:15.000000Z
      handler_name    : GoPro H.265
      encoder         : GoPro H.265 encoder
    Side data:
      displaymatrix: rotation of nan degrees
    Stream #0:2(eng): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 189 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:08:15.000000Z
    Stream #0:3(eng): Audio: pcm_s32le (in32 / 0x32336E69), 48000 Hz, 4.0, s32, 6144 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:08:15.000000Z
    Stream #0:4(eng): Data: bin_data (gpmd / 0x646D7067), 89 kb/s (default)
    Metadata:
      creation_time   : 2022-05-25T07:08:15.000000Z
      handler_name    : 	GoPro MET
    Stream #0:5(eng): Data: none (tmcd / 0x64636D74), 0 kb/s
    Metadata:
```

if reorder the stream of cliped file is rather hard, is it possible to clip the .360 file using ffmpeg? (reserve the cliped meta info)

dnewman-gpsw, 2022-12-11:
Not sure why the order matters.  

You can trim with ffmpeg, but it will lose the header metadata, which can be restored with udtacopy (https://github.com/gopro/labs/blob/master/docs/control/chapters/bin/udtacopy.zip)

e.g.
>ffmpeg -i D:\48HPF-Sync\Cam-Beach\GS010003.360 -map 0:0 -map 0:1 -map 0:2 -map 0:3 -map 0:4 -c copy -ss 00:00:05 -t 00:00:10 D:\48HPF-Sync\Cam-Beach\GS010003.trim.MOV
>udtacopy D:\48HPF-Sync\Cam-Beach\GS010003.360 D:\48HPF-Sync\Cam-Beach\GS010003.trim.MOV
>rename D:\48HPF-Sync\Cam-Beach\GS010003.trim.MOV D:\48HPF-Sync\Cam-Beach\GS010003.trim.360

closed 2022-12-11 by dnewman-gpsw

CanCanZeng, 2022-12-12:
Hi @dnewman-gpsw , thanks for your help!
I need the order to be the same because I use the following code to concat multi-segments
```
ffmpeg -y -f concat -safe 0 -i list.txt -c copy -map 0:0 -map 0:3 -an channel_0.mp4
ffmpeg -y -f concat -safe 0 -i list.txt -c copy -map 0:5 -map 0:3 -an channel_1.mp4
```
so I need the first video stream in channel 0 and second video stream in channel 5.

I tried your method, but the output *.trim.360 has different channels than the original 360 file. I find that the first step using ffmpeg to clip the videos cause lose of streams: the original file has 6 channels but the *.trim.MOV file has only 4 channels.