gopro/gpmf-parser #175: GoPro Max Very Wrong Magnetometer Readings

joshi-bharat, 2023-04-06:


dnewman-gpsw, 2023-04-07:
Little high, but it is working?   Rotate the camera through a short capture, is the data is good.

gbanfill, 2023-05-17:
Hi @dnewman-gpsw , I'm looking at the raw magnetometer readings too. I recorded 6 video clips in different directions (N, E, S, W Up and Down) the gopro was stationary while I recorded. I extracted the data below it seems reasonably stable. ( think these are μT). 

Direction| X | Y | Z
-- | -- | -- | --
North | 172 | 26 | 50
East | 158 | 15 | 50
South | 141 | 22 | 51
West | 145 | 34 | 50
Down | 195 | 26 | -12
Up | 90 | 30 | -11

According to wikipedia, the earths magnetic field is between 25 to 65 μT (https://en.wikipedia.org/wiki/Earth's_magnetic_field#Intensity) so these values seem a high?

What I really want to do is convert these values into Yaw, Pitch, Roll, but that's a different ticket.

dnewman-gpsw, 2023-05-17:
Just assume the data is not calibrated.  X has a bias in this data. Range 90 to 172, so subtract 131, and your range for x is -41 to 41.      Very crude, but it will get you data in the expected range.  

As for yaw, pitch and roll, calculations, that outside the scope of this project.  This is all about the retrieval of stored  sensor data, the calculated meaning of this data is something else.