gopro/gpmf-parser #181: HERO11 Mini missing GYRO and ACCL data

JuanIrache, 2023-08-08:
Hi. I ran into a H11 Mini sample where the accelerometer and gyroscope data are missing. Their fourCC is not even present in the data. All the other expected streams are there (SHUT, WBAL, WRGB, ISOE, YAVG, UNIF, SCEN, HUES, FACE, CORI...). Is this expected in any mode? Seems to happen with at least firmwares H22.03.02.20.00 and 2.30.

I can provide the data stream privately if necessary.

dnewman-gpsw, 2023-08-08:
That is not expected.  Any idea I how to reproduce this error?   CORI data without GYRO is weird, as CORI is based on GYRO data.  It happen more than once, as you mention to firmware versions?

JuanIrache, 2023-08-08:
I'm not sure how to reproduce it, sorry. I found this case on Facebook and was trying to help/investigate. The user claims his first video file had a few seconds of gyro data (but not for the entire video) and then all the subsequent files lack it, even after reinstalling different versions of the firmware. You can find a link to one of the affected files at the top of their FB post: https://www.facebook.com/groups/gopro.hero.group/posts/4104145329811935/

dnewman-gpsw, 2023-08-09:
IMU seems to have stalled or failed, as CORI data is also stuck at unity. User should contact support if the IMU isn't working.

closed 2023-08-09 by JuanIrache