gopro/gpmf-parser #183: GoPro HERO12 GPS

JuanIrache, 2023-09-06:
Hi. The lack of GPS data in HERO12 videos seems to be confirmed now. Is the GPS module physically not there? Or was it just disabled to save battery & reduce heat? If so, can we expect it to be activated at a later date via a Labs command?

Thank you

dnewman-gpsw, 2023-09-07:
There is no GPS hardware to my knowledge.

trackdaygenius, 2023-09-07:
OMG, I we hope this isn’t the case. 93% (40k plus) of our Motorsport users love the ‘All in one’ goodness of the GoPro, the built in GPS is the only reason they haven’t joined the Insta360 or DJI party…Someone at GoPro needs our data to rethink their Service & Product design strategy. Kofi - Team GWarmest RegardsTeam G - Discover your Genius with Smart Technology! ======TDG=====Track Day Genius © is owned and managed by Avant Garde Associates Ltd a company with limited liability, registered in England & Wales #3730200Apple App Store download:  https://appsto.re/gb/M3ZP3.iWebsite: http://www.trackdaygenius.comFacebook: https://www.facebook.com/trackdaygeniusInstagram: https://www.instagram.com/trackdaygeniusYouTube: http://www.youtube.com/user/TrackDayGeniusTwitter: https://twitter.com/TrackDayGeniusGoogle: https://plus.google.com/+Trackdaygenius

realchrisolin, 2023-09-07:
I know this isn't the place, but average customers without a social media presence (or are stuck in Twitter purgatory) rarely have the opportunity to relay feedback that gets heard.

> There is no GPS hardware to my knowledge.

I'm absolutely floored by this design choice. GNSS telemetry is generally what sets GoPro apart from it's competitors, save for limited exception like the 11 mini. I'll keep my more critical thoughts about marketing and product design of the 12 to myself, mainly because if I'm right, it'll be reflected in your quarterly sales figures compared to past years.

While on the topic, public comments from GoPro about this design choice mentioned that it was influenced by analytics data of how many customers were using the mobile GoPro app "stickers" feature. Most of your customer base using GNSS telemetry don't use GoPro's software to visualize it because the software you offer ranges from horrendous to useless on this topic and people like Juan have written software that eases this pain point. Basing such a critical product design choice off of statistical interpretation of analytics data like this is spectacularly short-sighted. If I had any stock in GoPro, I'd be selling it after this.

jenslar, 2023-09-08:
Removing the GPS from future models would be the worst possible decision GoPro could make in our case. I also need to add a warning next to my tools and warn users not to get a GoPro 12 (or probably not a GoPro at all...).

The lack of a GPS nullifies tools, workflows and research built on a small, rugged, camera that can log position. I have to warn every researcher around me not to get a GoPro, since synchronising position is key for our work, and having a single device that can do this is important in the kind of environments we operate in. While I guess GoPro owes its customers nothing, and I don't have numbers (what numbers though?), I still feel a bit betrayed to be honest.

It also means there's *absolutely no need to look in GoPro's direction when getting new cameras* since GPS is what set it apart from its competition.

I agree the GPS was sometimes unreliable (our old VIRB did better in some cases), but I could easily filter on gpsfix, and naively hoped it was in GoPro's interest to work on this. If a hypothetical external GPS module existed, perhaps not everything would be lost, but it would also be another device to carry and charge in already demanding environments.

Before this decision, GoPro was the only choice (since Garmin discontinued its VIRB series), now it may as well be the last. This is a sad day.

sjmarcy2, 2023-09-08:
Perhaps an even better arrangement is possible for the hard core action types such as myself?    Many of us have GPS GoPros AND external GPS loggers of some sort.    All of the devices have some combo of BT and or WiFi.    So maybe a link arrangement of some sort could be developed.   Add even bettersuper GPS from external to the latest GoPro footage.  I use external GPS good to 4 inches accuracy.     Plus with timecode sync I guess those with a few GoPros in use on a shoot can just make sure at least one is a 5-11 GoPro and sync via the timecode again in a manner to be developed (hopefully not manually).

JuanIrache, 2023-09-08:
Back in the day, it seemed like the gpmf format was designed to accept data from external devices. Here's a working example: https://github.com/gopro/gpmf-parser/blob/main/samples/hero6%2Bble.mp4 But I don't think there is documentation on how to do this. It would be very interesting now for GPS data. Even if it is not saved in a format Quik can read, other software options could.

jenslar, 2023-09-08:
While we're probably in the wrong "forum" for this (sorry for abusing 'issues'!), if GPMF would allow for custom data via external sensors (as the BLE example shows), there could be a (bulkier) way forward. Garmin FIT allows for custom data (e.g. third party product that logs data types not in FIT, such as my bike computer that can log data from external BT devices). This could perhaps be one way to go that also depends less on GoPro providing all the sensors, but instead put focus a flexible data format, and software. I'm just sad that with the Hero12 we can not hand over a single device and just tell participants to press record. Moving away from the multi-device data collection is what the action cameras allowed us to move away from.

realchrisolin, 2023-09-16:
Since others are investigating workarounds, I wanted to point out that the application Juan sells to visualize telemetry data will accept a GoPro video *and* a GPX file. I believe it will try to sync the telemetry data in the GPX file based on date/time of the file recorded by the camera and provides offset tools to manually correct inaccuracies.

I typically will sync GPX data from a Garmin tracker to video recorded with a H10 or 11 with a bad fix using Juan's application. Even if the GMPF track has a bad telemetry fix, the nature of GNSS relies on precise timing to calculate location using triangulation. GNSS satellites have onboard atomic clocks and even a single fix on a satellite should give me a relatively accurate GPS time for the GPX data to sync to. It might be harder on a 12 because you'll be relying on onboard date/time, but as long as it's regularly kept in sync with your phone's time, this should work. Even without a GPS receiver, the 12 is still logging other sensor/metadata to recorded videos.

jenslar, 2023-09-16:
They key with any external device is sync. I'm thinking any device that does Garmin FIT (ironic I guess), or logs to kml, gpx - whatever if fine format-wise. The external format isn't the issue, that's solvalble. When there's an on-board GPS module you can find a good spot just to get satellite sync and set datetime on the GoPro, that's now lost.

The beauty of an action camera with GPS is that you can set the datetime and use that across any data that's logged. These devices are great, all-in-one data hubs (with GPS) if your primary data is audiovisual.

I already have a FIT library, reading/writing KML, GPX etc is fine, but as you mention @realchrisolin we now have to rely on the time being correctly set on the GoPro. That's the issue. I'm kind of tired of devices being advertised as being designed for use in remote, "extreme" conditions, but feature wise (perhaps that should be "software wise") more tailored for the backyard user recording shorter videos. Then again, I'm perhaps being naive. The product has to sell.

The Hero11 Black introduced per-point datetime, and satellite lock level via GPS9 (and deprecated GPS5, which is still logged in parallel), which from a developer makes me extra salty. It seemed GoPro made an attempt to make the GPS data better, only drop the GPS module completely for their next product. I'm guessing there are some salty engineers at GoPro at this point as well.

jenslar, 2023-09-18:
Just to be a bit more constructive instead of ranting all the time (sorry about that... is there a dev-forum?):

Do we know any external device that can make use of the BLE funtioncality? I take it none currently exists? (any general recommendations for robust GPS-loggers?) If a slim, reliable device existed (from GoPro or third party) that could log directly to the GPMF stream that is perhaps preferable despite the extra battery to charge. Similar to Garmin's ANT protocol (all logged to the same FIT-file as the main device logs to). That would also mean we could add any sensor that obeys this protocol. Being naive again, it could even an opportunity for GoPro to accessorise further with various modules (a la DJI's cameras?) that could be powered by the camera (plug it into the usb-c slot or similar).

I also don't know if GoPro BLE has the bandwidth [or is even designed] for continuous logging (external accelerometer at 200Hz? 10Hz GPS?). 
GoPro's spec mostly talks about camera control which I assume would mostly be short, infrequent bursts: https://gopro.github.io/OpenGoPro/ble_2_0

KonradIT, 2023-09-18:
Actually yes, several Polar watches are able to send telemetry to GoPro HERO4/5/6, but not sure if the watches can connect to newer cameras or can send metadata to newer cameras. 

https://gopro.com/en/us/news/Polar-Showcases-New-GoPro-Connected-Products-at-CES

https://www.polar.com/blog/polar-v800-gopro-data-overlay-now/

Unfortunately this partnership was due to GoPro's now defunct developer program. COROS connects to GoPro cameras because of OpenGopro, so does Amazfit... maybe those 2 smartwatch manufacturers are inclined to also send telemetry data if GoPro documents how.

jenslar, 2023-09-18:
That Polar connectivity looks nice, indeed! Now that the Garmin VIRB has been discontinued for a while (which had ANT support), there's perhaps a renewed opportunity at third-party developer collaborations? Newer fitness watches can have great battery life even with GPS logging so they definitely won't die before the camera does (we usually run the cameras until the battery is drained - or *cough* shuts down from overheating).

jenslar, 2024-09-04:
It seems the GPS is back on Hero 13 Black? (https://gopro.com/en/ro/shop/cameras/learn/hero13black/CHDHX-131-master.html) If so, great to hear! It's not in the tech specs as far as I can see, but it's mentioned in the intro video, the on the main page together with new overlay, and also in the footnote regarding battery life ("GPS off"). @dnewman-gpsw Perhaps you could check with your colleagues if this can be added to the tech specs, since I suspect many of us frequenting this repo are looking exactly for this.

Also, if the GPS is indeed in there, can we assume that it's GPS9 only from now on? I recall something about GPS5 being deprecated with the Hero 11 Black which logged both.

sjmarcy2, 2024-09-04:
> It seems the GPS is back on Hero 13 Black? (https://gopro.com/en/ro/shop/cameras/learn/hero13black/CHDHX-131-master.html) If so, great to hear! It's not in the tech specs as far as I can see, but it's mentioned in the intro video, the on the main page together with new overlay, and also in the footnote regarding battery life ("GPS off"). ***@***.***(https://github.com/dnewman-gpsw) Perhaps you could check with your colleagues if this can be added to the tech specs, since I suspect many of us frequenting this repo are looking exactly for this.

Interest in the 13 and (possible) new Max just increased..

Sent with [Proton Mail](https://proton.me/) secure email.

realchrisolin, 2025-09-22:
@gopro deleting factual criticism that I predicted two years ago won't change the fact that your company is failing spectacularly and doesn't do you any favors retaining customers. With all due respect, you deserve to fail at this point and it's sad to watch. I won't be renewing my subscription nor will I be purchasing the 13 or Max 2.

dnewman-gpsw, 2025-09-22:
You are correct I should have closed is ages ago when it went off the rails, you random post 12 months later of an old thread, looked like trolling so deleted, lacking any technical value.  Thank you making it clear that this was no longer a technical discussion.