gopro/gpmf-parser #187: char with initial null, then characters

bradh, 2024-02-11:
I'm trying to do bit-equivalent reconstruction of the parsed date (to make sure the parsing and writing is valid).

I've hit a case with HERO 11 GPMF content in the `udta` has a run that looks like this:

` 0x52, 0x41, 0x54, 0x45, 0x63, 0x01, 0x00, 0x09, 0x00, 0x55, 0x54, 0x4f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00`

That (to me) says RATE, string, sample size 1, repeat 9, then its null terminated. However after that are some characters (U, T, O). Are those supposed to be ignored (just not zero'd out from some previous text, like "AUTO")?

dnewman-gpsw, 2024-02-11:
This is minor bug.  The camera firmware team like to keep structure sizes constant, as RATE could have a size of zero, and saving 12 bytes from the file.  The next best thing would be to zero all that data, but it seems they only zeroed the first byte. But some battles aren't worth it.  Just ignore. Type 'c' strings are null terminated strings (although some fields will not have stored the null.)

bradh, 2024-02-11:
Thanks. The logic I'm applying is:

if sample size is 1, then repeat is the maximum length of the (single) string;
else repeat is the number of strings, and sample size is the maximum length of each string.

The actual length of the string the maximum length or the null delimiter position, whichever comes first.

Anything after the null is irrelevant, and those make for sane test cases.

closed 2024-02-17 by bradh