gopro/gpmf-parser #191: Understanding ACCL and GYRO orientation

henriberisha, 2024-05-25:
Hello GoPro team 😊

First of all i want to say thank you for your amazing products and customer service. 

I need some help understanding the orientation for ACCL and GYRO. 

I am using a Hero11 Black with SW V2.30
I am extracting some of the GPMF data and i am getting these values: 
```
('DEVC', 'null', 1, 6512)  --->>> nested below
	('DVID', 'L', 4, 1)  --->>> 1
	('DVNM', 'c', 12, 1)  --->>> HERO11 Black
	('STRM', 'null', 1, 1160)  --->>> nested below
		('STMP', 'J', 8, 1)  --->>> 10050065
		('TSMP', 'L', 4, 1)  --->>> 2198
		('STNM', 'c', 13, 1)  --->>> Accelerometer
		**('ORIN', 'c', 3, 1)  --->>> 'ZXY'**
		('SIUN', 'c', 4, 1)  --->>> 'm/s²'
		('SCAL', 's', 2, 1)  --->>> 417
		('TMPC', 'f', 4, 1)  --->>> 31.693359375
		('ACCL', 's', 6, 175)  --->>> [[3827, -97, 354], [3824, -103, 370], [3835, -119, 364], [3859, -123, 355],...............]
```

My confusion starts here as the 'ORIN' says **ZXY** while the documentation [GPMF](https://gopro.github.io/gpmf-parser/) says **Y, -X, Z**

In case it is Y, -X, Z; does this mean that this value [3835, -119, 364] breaks down into:
Y = 3835
X = 119
Z = 364   ?

Is this the same for ACCL and GYRO?

Thank you very much

dnewman-gpsw, 2024-05-27:
I don't see where the docs state Y, -X, Z, and if it did, it would only be an example.  Earlier cameras used different orientations, and ORIN and ORIO described how the data was stored and how it was to be decoded.  But the last few years all cameras have used ZXY orientation for both ACCL and GYRO.

henriberisha, 2024-05-28:
Thank you for your quick reply and clarification. I appreciate that. makes a lot of sense.
I was reading the GPMF doc and there are tables that describe different FOURCC. It starts with HERO5 and Session in which data order is ZXY then Fusion changes to -Y,X,Z, then HERO6 Black changes to Y,-X, Z. From here on to the newer versions there are no changes, so i was following the logic that newer versions support the previous versions metadata if there are no changes.

dnewman-gpsw, 2024-05-28:
I see. When a camera specifies the data order use that, the older cameras didn't provide ORIN.

henriberisha, 2024-05-29:
Thank you very much for your support. I am closing this ticket

closed 2024-05-29 by henriberisha