gopro/gpmf-parser #194: GoPro Hero 13 Black GPS

jenslar, 2024-09-04:
It seems the GPS is back on Hero 13 Black? (https://gopro.com/en/ro/shop/cameras/learn/hero13black/CHDHX-131-master.html)

If the GPS is indeed in there, can we assume that it's GPS9 only from now on? I recall something about GPS5 being deprecated with the Hero 11 Black which logged both.

If so, great to hear! It's not in the tech specs as far as I can see, but it's mentioned in the intro video, the on the main page together with new overlay, and also in the footnote regarding battery life ("GPS off"). @dnewman-gpsw Perhaps you could check with your colleagues if this can be added to the tech specs, since I suspect many of us frequenting this repo are looking exactly for this.

blade2005, 2024-09-04:
try this
https://gofile.io/d/yMHabA
select the desired compiler in the installer

dnewman-gpsw, 2024-09-09:
GPS is back with a GPS9 track in the GPMF track.  I just updated the repo's readmes.

jenslar, 2024-09-10:
That's great, thank you!

closed 2024-10-01 by dnewman-gpsw