gopro/gpmf-parser #195: Extract all metadata but in intervals of video seconds

MarioCavero, 2024-10-01:
I want to extract metadata of a gopro9 video but in defined intervals of the inputted video, f.g, [8,15,23], a video of 30 secs, should output me instead of a complete metadata, the metadata splitted in "video intervals", 0-8, 8-15, 15-23, 23-30. 
I can investigate doing it manually knowing the Hz of each sensor (including video) and calculating how much data I should be taking into account having in mind  video fps (Ref: https://github.com/gopro/gpmf-parser/issues/118) and also the chunks in which the telemetry comes. Is there any kind of more straight forward way?

dnewman-gpsw, 2024-10-01:
The metadata is clumped into video GOP lengths. For most 60Hz based systems, 30/60 frames per metadata payload. This approach will not get you the same precision of computing the sample rate for each sensor.  It works fine for lower frequency data like GPS and CORI, but GYRO and ACCL, benefits from extracting it all.

MarioCavero, 2024-10-02:
Sounds good, thank you very much for the quick response @dnewman-gpsw . If I have any future issues I will open (another issue) or re-open. I will close the issue for now.

closed 2024-10-02 by MarioCavero