gopro/gpmf-parser #196: What exactly is MXCF and MAPX?

walt-grace, 2024-11-04:
I am trying to undistort the HyperView image. I found these metadata in gpmf, feels like the calibration data. 

MXCF: x1x3x5x7x9x11x13x1y2
MAPX: 1.5805143, -8.1668825, 74.519875, -451.50024, 1551.2922, -2735.5422, 1923.1572, -0.10860265

Can you please explain how to interpret this data? Thanks

dnewman-gpsw, 2024-11-05:
Not calibration, but a distortion formulas.  You have part of it.

    "MXCF": ["x1", "x3", "x5", "x7", "x9", "x11", "x13", "x1y2"],
    "MAPX": [1.5805143, -8.1668825, 74.5198746, -451.5002441, 1551.2922363, -2735.5422363, 1923.1572266, -0.1086027 ],
    "MYCF": ["y1", "y3", "y1x2", "y1x4"],
    "MAPY": [1.0238225, -0.1025671, -0.2639930, 0.2979266 ],
    "PYCF": ["r0", "r1", "r2", "r3", "r4", "r5", "r6"],
    "POLY": [0.0000000, 1.8229533, 0.1068096, -0.6631866, 0.3563110, -0.0000000, 0.0000000 ],
    "ZMPL": 0.6549,
    "ARUW": 1.1429,
    "ARWA": 1.7778

The radial distortion  (from POLY and PYCF)
angle = 1.8229533r + 0.1068096r^2 - 0.6631866r^3 + 0.3563110r^4

Same technique for the 2D distortion for HyperView

(from MAPY and MYCF)
y_out = 1.0238225y - 0.1025671y^3 -0.2639930yx^2 + 0.2979266yx^4

(from MAPX and MXCF) "x3", "x5", "x7", "x9", "x11", "x13", "x1y2
x_out = 1.5805143x - 8.1668825x^3 + 74.5198746x^5 - 451.5002441x^7 + 1551.2922363x^9 - 2735.5422363x^11 + 1923.1572266x^13 - 0.1086027xy^2

This is not easy.  But all the math used is provided.

walt-grace, 2024-11-05:
Thanks for clarifying my confusion.

What about the camera intrinsic? I see the HyperView 35mm Equivalent Focal Length is 12mm. Can I estimate it like:

35mm film is 5600x3620(36mmx24mm). Final out size is 3840x2160
fx = fy = 12/(sqrt(36^2+24^2)/sqrt(5600^2+3620^2)) = 1849 pixel
Cx = 1920, Cy=1080

dnewman-gpsw, 2024-11-05:
I don't know if the market equivalent focal length are real.  As the distortion profile is different, particularly for HyperView, I wouldn't trust 12mm has any meaning outside of linear projections (is 12mm HFOV, VFOV or ZFOV equivalent?) The ZFOV metadata is the diagonal FOV, this can be used to determine a 35mm equivalent (when ignoring distortion.) 

If I shot 16x9 Linear with HS on, ZFOV is 100.53°.  Which is about 17mm full frame (at 16x9) equivalent.  I used https://www.scantips.com/lights/fieldofview.html