gopro/gpmf-parser #198: gpmfdemo crashes with segmentation fault

gezabohus, 2025-02-23:
Hey,

I built gpmfdemo as per the instructions on an iMac (Ventura 13.7.3) without errors and warnings, but I get:

demo % ./gpmfdemo /tmp/GX010084.MP4 -g
zsh: segmentation fault  ./gpmfdemo /tmp/GX010084.MP4 -g

The video exists and is readable.

Please advise.

PS. Just in case:

demo % gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin22.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

dnewman-gpsw, 2025-02-24:
Could you test with one of the sample mp4s included in the Repo?  https://github.com/gopro/gpmf-parser/tree/main/samples  If it fails on those, that would be puzzling as nothing has changed in years. If it is something in the new media, I will need to know what that is, or and maybe get a sample.

dnewman-gpsw, 2025-02-27:
I can't get it to fail.  I've tried Linux and Windows. 

(base) david@DAN12900KS:/mnt/z/code/gpmf-parser/demo$ make
gcc -g -c GPMF_demo.c
gcc -g -c ../GPMF_parser.c
gcc -g -c ../GPMF_utils.c
gcc -g -c GPMF_mp4reader.c
gcc -g -c GPMF_print.c
gcc -o gpmfdemo GPMF_demo.o GPMF_parser.o GPMF_utils.o GPMF_mp4reader.o GPMF_print.o -lasan
(base) david@DAN12900KS:/mnt/z/code/gpmf-parser/demo$ ./gpmfdemo ../samples/hero8a.mp4 -g
VIDEO FRAMERATE:
  29.970 with 379 frames
PAYLOAD TIME:
  0.000 to 1.001 seconds
GPMF STRUCTURE:
  DEVC nest size 6660
    DVID type 'L' size 4 data: 1,
    DVNM type 'c' size 11 data: "HERO8 Black"
    STRM nest size 1376
 ... 

Seems the issue might be local to you, can you debug where it is crashing?

gezabohus, 2025-02-27:
Hey, thanks for spending time on this. I haven't done c/c++ in a long time, but I'll try to take a look. I'll get back to you.