gopro/gpmf-parser #199: Chaptered videos

p98d, 2025-03-03:
I am not being able to extract the data from chaptered videos. My recordings are around 1h long and they therefore get split in chapters by my Hero10 and Hero13. When I try to run the demo I am only able to access the data stream from the last chapter of each video.

Is there a way to either access the timestamps and GPS data from the single videos or to merge them in a way that keeps all metadata intact and readable? [mp4-merge](https://github.com/gyroflow/mp4-merge) looks promising, but when I try to run the demo on the joined file I still get the "is an invalid MP4/MOV or it has no GPMF data" error.

p98d, 2025-03-03:
I am trying on all the 5 chapters of the video I took with my hero10. I get the correct output for the last chapter, but for all the previous one I get something looking like this (with PRINT_MP4_STRUCTURE):
```
 ftyp (20)
 free (8)
 mdat (3798176462)
error: Hero10d.mp4 is an invalid MP4/MOV or it has no GPMF data
```

p98d, 2025-03-04:
I was just trying to figure out why it would work with some files and not with others, I really do not see what I am doing wrong here.
I can share the videos privately to try and reproduce the issue, though.

dnewman-gpsw, 2025-03-04:
Those files are all working perfectly here using the current version of this repo.  Even the joined media is working correctly. This error is local to your development environment.  I have now reconfirmed, with your footage, on Windows, Linux, and MacOS. I expect it is because you are using some older libraries or system that is defaulting to using a 32-bit fseeko() / ftello().  This will impact any files greater than 2GB.  I have created a branch called force64. In this branch I set `#define _FILE_OFFSET_BITS 64` that is supposed to help with this issue.

```
git pull
git checkout force64
make
```

See if that solves the issue.

p98d, 2025-03-05:
That worked! I had not realized it was a 32-64bit issue, thank you very much!

closed 2025-03-05 by p98d