gopro/gpmf-parser #203: Segmentation fault in OpenMP4Source

yangzao, 2025-09-13:
Hi, when I was testing GPMF-parser (latest commit built on x86-64 Ubuntu 22.04) with random inputs, I found one causing segmentation fault in `OpenMP4Source`.

[1.zip](https://github.com/user-attachments/files/22315269/1.zip)

Command line used:
```
./gpmfdemo ./1 -g -a -i -s -c -v -t
```

Crash line: `gpmf-parser/demo/GPMF_mp4reader.c:780`

```
"--->780    	fileoffset = (uint64_t)metaoffsets64[stco_pos];",
```

Back trace:
```
"#0 0x55555557cced in OpenMP4Source gpmf-parser/demo/GPMF_mp4reader.c:780",
"#1 0x555555555cef in readMP4File gpmf-parser/demo/GPMF_demo.c:211",
"#2 0x555555555960 in main gpmf-parser/demo/GPMF_demo.c:133",
```

Let me know if you need more information!

yangzao, 2026-01-15:
I just tried it on Windows and it still crashes

[image]

dnewman-gpsw, 2026-01-15:
No crash in with Window Visual Studio build.

dnewman-gpsw, 2026-01-15:
Now fixed with the v2.0.3 of GPMF_mp4reader.c

closed 2026-01-15 by dnewman-gpsw

yangzao, 2026-01-15:
> Now fixed with the v2.0.3 of GPMF_mp4reader.c

Great, thank you for confirming the bug!